krautech/btt-eddy-guide #5: Removing KAMP not 100% necessary. Adaptive Purge still works

thomasfjen, 2024-05-13:
[image]

You mention in your guide that KAMP should be removed. That's not entirely correct, you simply shouldn't use the meshing part anymore and rely on the klipper adaptive one, like you mentioned. 

You still can use the purge part without any problems. 

[image]

krautech, 2024-05-13:
You are indeed correct, I shall update this now.

krautech, 2024-05-13:
You are indeed correct, I will update this now, thank you

krautech, 2024-05-13:
The guide has been updated :) thank you!

closed 2024-05-13 by krautech