krautech/btt-eddy-guide #6: No valid measurements found at coordinate

tonywright, 2024-05-19:
Hi, I have followed your excellent guide to install the eddy probe on my klipper printer but when doing a mesh (Rapid or detail) it always fails with "No valid measurements found at coordinate   [36.0, 60.0] 
I have reduced the bed mesh size all the way down to just a small 100mmx100mm mesh in various places on the bed and I still get the error, all be it with different numbers.

Another issue, although not as important is when using the Height Mesh Calibrate function, I get "Must calibrate probe_eddy_current first" (Which I have obviously done)
It is working fine as a Z end stop.

Any ideas on the first issue?

Thanks

Tony

krautech, 2024-05-19:
Interesting..

Put your toolhead at those coordinates

X36 Y60

Where does eddy sit? Still over the build area and not off at all?

Second issue you should only be getting if you haven't done the
LDC_CALIBRATION steps. But you wouldn't be able to do a bed mesh if you
hadn't.

I wonder if it could be a firmware issue? Perhaps giving a reinstall of the
firmware a try if you haven't already.

tonywright, 2024-05-19:
Hi, yes the eddy is over the build plate, I also moved the mesh a few  different places over the bed in a 100x100mm grid and I got the same error but with different numbers so I don't think it is the x/y that is a problem.
I have also now re-flashed it and still have the same issue. 
When you generate the image for it do you leave the USB ID's as they are? (In MenuConfig)
The ldc has definitely been done as it has stored the value (15) in Printer.cfg

I did see this issue (The only result from google) but on a German forum but it had no answer.

So annoying as I am so close to getting it working.
Mind you the adaptive mesh doesn't work or at least for stl's sliced with Cura.

krautech, 2024-05-19:
That's incredibly strange I haven't seen this issue pop up anywhere yet.

I leave the USB stuff alone in make menuconfig.

Hmm,

Make an issue over on https://github.com/bigtreetech/Eddy/issues as that's
where the bulk of things are going and bound to be seen by BTT.

I'll try discuss this with some people today and see what we come up with.

It's basically like everything is ignoring Eddy's data and saying nothing
is coming from Eddy even though everything is set correctly?

Atleast that's how it seems.

It looks like you've got everything right though, but there could be some
conflicts with other things.

Could you post your printer.cfg?

Thanks

tonywright, 2024-05-20:
[config-20240520-071400.zip](https://github.com/krautech/btt-eddy-guide/files/15374177/config-20240520-071400.zip)

printer.cfg zipped and attached.
 I did have a thought, I haven't managed to do a temperature drift calibration but didn't think this was required as it is only available on the usb version. You don't think that could make a difference do you?
I have checked the klipper log and it does show the output of each reading and they are all "Invalid".
Also I did check and the z homing is not done with the probe, when I tried to use it I just got the "Eddy needs calibrating" error.
I will try to do another temp drift calibration again. I keep getting an overlap error, but again the message for that is odd as it says overlap at point 54c but I am only testing up to 47c

krautech, 2024-05-20:
If you're using eddy coil skip the thermal drift calibration, it's unable
to process it.

I'd remove any of the config it added and start again but skipping that

krautech, 2024-05-20:
Printer.cfg looks fine to me.

What happens when you start the z offset calibration?

https://github.com/krautech/btt-eddy-guide?tab=readme-ov-file#3-z-offset-calibration

tonywright, 2024-05-20:
I am sure I did this the first time and didn't have a problem, just did it again and first run had to turn the extruder temp up, then second time had to increase extrude max distance, third time lucky, after ejecting the filament (There was none in it) it worked. Just now tried a mesh and it worked! I am so embarrassed that I must have missed this. Sorry to waste your time but I guess the next time another idiot gets this error we will know what to suggest.

Tony

closed 2024-05-25 by krautech