krautech/btt-eddy-guide #7: Z-Offset workaround.

ifreislich, 2024-05-27:
After struggling for a day or so with Z-Offset not being saved I've arrived at the following procedure which works reliably:

1. Home all axes, no need to heat the bed or nozzle.
2. Move the nozzle to the XY coordinates that Eddy sensor was during homing.
3. PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy
4. Do not do the paper test. In stead adjust the Z value to zero
5. Now apply the Z-Offset so the manual probe value reads your Z-Offset.
6. Accept and save once once the calibration is complete.

Print a single layer and measure the printed layer thickness with calipers. Use the difference between the intended layer height and measured layer to adjust the eddy calibration as above.

krautech, 2024-05-27:
Howdy, 

Someone else attempted this, the problem becomes that once you apply heat and thermal expansion is applied to your printer, the offset could be wrong which will result in a nozzle crash. Thermal drift calibration is supposed to counter this but has shown to be unreliable.

You could attempt this at your normal printing temperatures and even then, if you change build plate or temperatures you could have a varied z-offset. 

Thermal Drift needs to be worked on and hopefully will have some changes soon to better counter this.

Fingers crossed anyway.

ifreislich, 2024-05-27:
The thermal calibration takes me about 1.5 hours to complete. I think that the key is to do the thermal calibration at the homing location. I use the CLI to adjust things during the calibration procedure. BTT says to use the center of the bed but the center of my bed moves around too much as the thermals change. I home above my reference bed screw and rely on adaptive mesh for the rest.

The bed coordinates of my Z home probe are X23.3 Y15 so I do the following:
1. `echo "G1 X135 Y135 F6000" >printer_data/comms/klippy.serial` place the Eddy in the center of the bed
3. slowly ramp the bed temperature until the manual probe is triggered.
4. `echo "SET_HEATER_TEMPERATURE HEATER=heater_bed TARGET=XX" >printer_data/comms/klippy.serial` where XX is 2C to 5C lower than the bed setpoint that caused the manual probe at the node so that the Eddy stabilizes at the node temperature, multiple adjustments may be necessary.
5. `echo "G1 X23.3 Y15 F6000" >printer_data/comms/klippy.serial` move the nozzle to the home probe position.
6. perform the manual probe "paper test" ensuring that the paper resistance is consistent been each node and drop the nozzle by the paper thickness (0.12mm in case)
7. accept the position.

repeat until all temperature nodes have been probed.

I've just done a print test with the bed at 100C (normally 60C) and the Eddy heat soaked to 62C (normally 35C to 42C). My first layer slicer height is 0.25mm, measured height 0.27mm.

First layer results:
Bed 60C,   Eddy 38C: first layer +0.01mm
Bed 100C, Eddy 62C: first layer +0.02mm

closed 2024-05-31 by ifreislich