krautech/btt-eddy-guide #9: eddy in klipper main?

tonywright, 2024-07-09:
Hi, I notice that you updated the readme 9 hours ago but the klipper main now supports the eddy. At least it is running on my printer with a couple of amendments, ie data_rate not required (And fails with it in)
and the bed_mesh_calibrate mode is "Rapid_Scan" now not just "Rapid"
Also the "Temperature_Probe" needs to be "Temperature_sensor"

I did have an issue with klipper failing when doing the "PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy" command but it worked ok when I rem'd out the second extruder/stepper on my IDEX printer. Putting it back after calibrating worked ok then.

Tony

krautech, 2024-07-09:
Yeah this wasn't a full update it's just to pull it back in line since we stopped using this repo and now use the BTT eddy repo to manage this guide.

Klippers implementation is still missing key functionality for BTT eddy and there is currently a pull request waiting to finalise it.

So once it's fully implemented and working across the board, this guide and BTT eddy guide will both be fully updated.

texidor3d, 2024-08-05:
majority of the functionality of the eddy probe has been been merged into mainline klipper. The only option that doesnt work is the data_rate option. My probe functions but the mesh is wavy. What am i missing?

krautech, 2024-08-05:
Mainline klipper is still missing alot of the functionality actually. BTT is still waiting on a merge of their PR to get things finished. 

Wavy mesh usually means you have interference. Could be your cable or some kinda of electronic interference. Are you doing a bed mesh while your bed is heating? Or cold?

texidor3d, 2024-08-05:
I've performed a scan at both room temperature and fully heated at 210
hotend and 60 bed. The wire is stock length routed far away from any
stepper motor wires. I do however have aluminum chimney tape I'm using as a
shim on top of the magnet in-between the heated bed and metal plate

krautech, 2024-08-05:
I'd give it a try without the aluminium and see how the scan results go