krautech/btt-eddy-guide #10: Nozzle crash after performing Z_TILT_ADJUST with the Eddy coil

cssi-dev, 2024-08-19:
I tried printing the first part after switching to the new probe. I was suprised to learn, that my print was canceled during the print start macro.
Error: 
> probe triggered prior to movement

with `horizontal_move_z: 5 `

So I switched it to `horizontal_move_z: 10`(even though In the bed section the horizontal_move_z is just 2 ), to then get the nozzle crash because the printer after getting all the readings moved all z axes way to far.

I get that you are not supposed to use the eddy coil for homing because of the temprature drift. But please let us use the Tilt adjust. I don't want to switch every 5-10 prints between the probes just to readjust the z axis when it gets out of wack.

Steps to reproduce:
1. Home the Printer
2. Start `Z_TILT_ADJUST`
3. Be ready to shutdown the printer befor something bad happens
[klippy.log](https://github.com/user-attachments/files/16666768/klippy.log)

krautech, 2024-08-19:
Please refer to 

[BTT Eddy](https://github.com/bigtreetech/Eddy)

For up to date information

cssi-dev, 2024-08-20:
You mean move this Issue to the other repo?

krautech, 2024-08-20:
Yeah it'll get more attention over there.

This repo was setup when I first built the guide before it was adopted by BTT. It's maintained on their official repo now.

Instructions etc have changed which could also solve some issues

cssi-dev, 2024-08-20:
Ok, thank you

closed 2024-08-20 by cssi-dev