krautech/btt-eddy-guide #14: Failed to flash

DecoderRingClub, 2024-11-16:
I'm currently unable to flash the eddy properly on the mainline version of Klipper. I've run the following commands:

1. cd ~/klipper
2. make menuconfig

[image]

3. make
4. disconnected power and held the reset button before connecting power
5. lsusb
6. Found Bus 001 Device 005: ID 2e8a:0003 Raspberry Pi RP2 Boot
7. make flash FLASH_DEVICE=2e8a:0003

but I get the error:

```
Flashing out/klipper.uf2 to 2e8a:0003
sudo lib/rp2040_flash/rp2040_flash out/klipper.uf2

Loaded UF2 image with 149 pages
Found rp2040 device on USB bus 1 address 5
Flashing...
Resetting interface
Locking
Exiting XIP mode
Erasing
Flashing
Rebooting device
Command reboot failed with status 1: unknown command

Failed to flash to 2e8a:0003: Error running rp2040_flash

If the device is already in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=2e8a:0003

Alternatively, one can flash rp2040 boards like the Pico by manually
entering bootloader mode(hold bootsel button during powerup), mount the
device as a usb drive, and copy klipper.uf2 to the device.

make: *** [src/rp2040/Makefile:78: flash] Error 255
```

Quinemc, 2024-11-22:
Having the exact same issue too. Have you found the fix?

DecoderRingClub, 2024-11-22:
> Having the exact same issue too. Have you found the fix?

I mounted the USB as a drive and moved over the firmware. Then, power cycled it to reset it/flash the firmware. The Eddy populated under `ls /dev/serial/by-id/*` after. I don't know if that is the best way, but nothing else worked, and it is a method.

krautech, 2024-11-22:
> I mounted the USB as a drive and moved over the firmware. Then, power cycled it to reset it/flash the firmware. The Eddy populated under `ls /dev/serial/by-id/*` after. I don't know if that is the [...]

This is the preferred method for initial firmware updating. You can also flash Katapult and use Katapult to flash Klipper after you've done that

Nevio024, 2024-11-23:
Can someone go further into this? I'm having the same issue but don't know how to mount it as a USB and pull the firmware over...

I tried to connect it to my windows laptop, but it doesn't get recognized...

Would be really glad if someone yould help!